ryanbinns/ttwatch #109: ttwatchd does not work

ladywithhat, 2016-12-25:
With the update of ttwatch my ttwatchd deamon does not work anymore. Before that it worked just fine.
Files do not get pulled automatically of the watch and to use ttwatch i have to kill ttwatchd.
How to fix ttwatchd?
Current setup:
**/etc/systemd/system/ttwatchdaemon.service**
`[Unit]
Description=Starts ttwatchd for tomtom sport gps watch to sync activity files automatically

[Service]
Type=forking
ExecStart=/usr/bin/ttwatchd --runas user:usb --get-activities --update-gps --update-fw --set-time -s /home/user/activities

[Install]
WantedBy=graphical.target`

ryanbinns, 2016-12-28:
None of the files used by `ttwatchd` have changed for over 2 months, so the latest update cannot have broken it. The last 2 updates have only changed the main file for the `ttwatch` program. Have you tried deleting your build directory and doing a full rebuild?

bioxz, 2017-03-05:
Did you check your log? I have more or less the same problem. Daemon stopped working and I have to kill it to access the watch.  The log says
```
Sun Mar  5 14:39:40 2017: Watch connected...
Sun Mar  5 14:39:40 2017: Watch not processed - does not match user selection
```

There is no config file and I'm running it with

`/usr/local/bin/ttwatchd --update-gps --set-time --runas=bioxz:tomtom`

Everything besides the daemon works fine (as long as the daemon isn't running), I can update the GPS running ttwatch as a user. With the daemon running, there isn't even any output from --devices. Running the daemon without the -d should process any watch, right?

ryanbinns, 2017-03-06:
I haven't got that error, but I did notice today that the daemon wasn't working. It was broken by a pull request in January, well after this issue was opened. I have fixed it, so have another go.

Yes, running without `-d` will process any watch. You should still be able to run `ttwatch`, even with the daemon running, but bear in mind that the daemon will grab the watch as soon as it's plugged in, so you'll have to wait for it to finish with the watch before trying to use `ttwatch` to do other things.

I have noticed that the most recent firmware doesn't seem to report USB configuration strings correctly, so the device serial number is not available, which makes the device filter not work properly. You should still be able to filter by watch name though. Running `lsusb -vvd1390:` shows errors for the configuration strings, which used to work perfectly, so I think they have introduced a bug in the firmware recently.

ecorini, 2018-06-24:
Also for me the daemon is not working properly.
If i start the daemon with the watch already connected, the watch is recognized:
```
Sun Jun 24 21:37:37 2018: Watch connected...
Sun Jun 24 21:37:40 2018: Downloading GPSQuickFix data file...
Sun Jun 24 21:37:40 2018: Writing file to watch...
Sun Jun 24 21:37:40 2018: GPSQuickFix data updated
Sun Jun 24 21:37:41 2018: Current firmware is already at latest version
Sun Jun 24 21:37:41 2018: Finished watch
```

but if i connect later:
```
Sun Jun 24 21:38:08 2018: Watch connected...
Sun Jun 24 21:38:08 2018: Watch not processed - does not match user selection
```
The `ttwatch` command works correctly.
It's possible that the device filter wrongly analyze the music drive, instead of the workout drive?
Device is tomtom Adventurer, productID 0x7475 with latest firmware: 1.7.64
Here is the ouput of the `lsusb -vvd1390:` cmd:
```
Bus 001 Device 019: ID 1390:7475 TOMTOM B.V. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1390 TOMTOM B.V.
  idProduct          0x7475 
  bcdDevice            1.00
  iManufacturer           1 TomTom
  iProduct                2 TomTom GPS Music Watch
  iSerial                 3 HL4517I00012
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           64
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              4 Workout Drive
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      36
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0100  1x 256 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0100  1x 256 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              5 GPS Watch Music Drive
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  bNumConfigurations      1
Device Status:     0x0001
  Self Powered
```

ecorini, 2018-06-25:
I enabled libusb debug putting `libusb_set_debug (NULL, 4);` in `main` and when the watch is hot-plugged i get this error:
```
[15.170678] [000009c8] libusb: debug [udev_hotplug_event] udev hotplug event. action: bind.
[15.170682] [000009c8] libusb: error [udev_hotplug_event] ignoring udev action bind
```
the error code of the function `ttwatch_open_device` in hotplug callback is `TTWATCH_UnableToSendPacket`.

ecorini, 2018-07-21:
seems to be a problem related to libusb: with version 1.0.17 daemon is working, but with version 1.0.21 (default in ubuntu 18.04) is not.

ryanbinns, 2018-12-04:
Sorry for taking so long to get to this. This was unfortunately caused by me failing to read the documentation on hot-plug events properly. I was calling synchronous libusb functions from within the hotplug callback handler, which is prohibited. It worked in previous verions of libusb, but not in later ones. I have updated the code to do this properly, and it works properly on my system. Hopefully it will fix your issues.

closed 2018-12-04 by ryanbinns