ryanbinns/ttwatch #116: Firmware Updates

frankcrawford, 2017-02-05:
TomTom seem to bring out firmware updates very regularly, and as the current fix involves modifying ttwatch.c and recompiling, which may concern some users, any chance of enabling a default setting?

The most likely default would be the last entry in the MANIFEST_DEFINITIONS table, since that is what most people do.  If this is considered too dangerous to do automatically, then at least it could be an option to enable it.  In all cases it should still print out that the firmware version isn't found.

What are your thoughts?

ryanbinns, 2017-02-08:
I've thought about this before, but thought it would be too dangerous. I could possibly allow the program to *read* settings, but not *write* them if the firmware version was not found. Perhaps a "force" option could be added if people really wanted to write settings.

frankcrawford, 2017-02-08:
Yep, understandable, and yes, read-only would be a reasonable solution.

ryanbinns, 2017-02-10:
I have implemented the ability to read settings if the firmware version has not been found. It still prints out that the firmware version has not been found. A `--force` option has been added to allow writing them as well, if people are feeling brave...

closed 2017-02-10 by ryanbinns