ryanbinns/ttwatch #118: TomTom Adventurer not supported

scubajorgen, 2017-03-06:
The newest GPS watch is not supported. The Adventurer is equivalent to the Runner but has more activities (trailrunning, skiing, etc) and measures height based on barometric pressure.

ryanbinns, 2017-03-07:
Yep. I don't have one, so I can't really add support for it unless someone wants to reverse-engineer it for me. Altitude shouldn't be an issue as I assume it probably reports altitude similarly to the Multisport. I would need ttbin and tcx/gpx/csv/kml files generated by the TomTom software to be able to add support myself. If someone wants to provide these I'm happy to do the work.

scubajorgen, 2017-03-07:
Thanx for the good work.
I forked your code and added the new activities.
I found out about the altitude record and will add it. If you are 
interested I can give you access,  but i am not really familiar with git yet...

Jorgen

ryanbinns, 2017-03-09:
If you would like your code added, create a pull request. Just go to the repository top-level on github and click "New Pull Request". It's fairly straightforward.

closed 2017-03-10 by ryanbinns