ryanbinns/ttwatch #119: ttwatch ttbin not compliant to tomtom ttbin

scubajorgen, 2017-03-06:
When I compare the ttbin file generated with ttwatch to the ttbin file downloaded with TomTom Sports, I notice a difference: the timestamp in the TAG_STATUS record (0x21) in the TomTom version seems to be in local time, whereas the timestamp in the ttwatch file appears to be GMT.
In the file ttbin.c the timestamp is converted by ttwatch during parsing of the ttbin data by subtracting the utc_offset (parse_ttbin_data()).
I noticed by removing the file deletion after download in ttwatch, downloading the file using ttwatch, then downloading (and erasing) the same file with TomTom Sports

ryanbinns, 2017-03-07:
Interesting. `write_ttbin_file` reverses the subtraction by adding the UTC offset back in before writing the file, so the timestamps should be identical. Are you able to attach both copies of a ttbin file so I can investigate?

scubajorgen, 2017-03-07:
I'll send you a couple of files

Jorgen

scubajorgen, 2017-03-08:
The function write_ttbin_file() is never used to write the ttbin file. The function do_get_activities_callback() reads the 'raw' ttbin data in *data. The ttbin data is parsed (parse_ttbin_data()). En passant is the 'raw' data modified. After parsing the raw data is written to file.
The solution is not applying the calculation to the raw data:
p.record->status.timestamp -= file->utc_offset;
but modifying the parsed data (after parsing): 
record->status.timestamp -= file->utc_offset;

[example.zip](https://github.com/ryanbinns/ttwatch/files/828860/example.zip)

ryanbinns, 2017-03-09:
Yes, you're right. I have fixed the code. It was wrong for several other record types as well.

closed 2017-03-09 by ryanbinns