ryanbinns/ttwatch #121: ttwatch does not find new device

vmindru, 2017-03-16:
Hi i can not perform initial setup on tomtom touch 

```
[vmindru@zetor ~]$ ttwatch --initial-setup
Unable to open watch
[vmindru@zetor ~]$ ttwatch --devices
[vmindru@zetor ~]$ 
```
```
[vmindru@zetor ~]$ lsusb | grep -i tom
Bus 003 Device 005: ID 1390:7480 TOMTOM B.V. 
[vmindru@zetor ~]$ 
```

ryanbinns, 2017-03-17:
Does it work if you run `sudo ttwatch --devices`?

vmindru, 2017-03-17:
nope =( not a perms issue

vmindru, 2017-03-17:
I will try to install this now on my laptop. let's see how this goes.

vmindru, 2017-03-17:
same problem on Fedora 23

vmindru, 2017-03-17:
ok i got it partially working with 

```
[vmindru@vmindru ttwatch]# git diff src/libttwatch.cpp
diff --git a/src/libttwatch.cpp b/src/libttwatch.cpp
index 985d2de..037914f 100644
--- a/src/libttwatch.cpp
+++ b/src/libttwatch.cpp
@@ -354,9 +354,9 @@ int ttwatch_open_device(libusb_device *device, const char *serial_or_name, TTWAT
 
     // ignore any non-TomTom devices
     // PID 0x7474 is Multisport and Multisport Cardio
-    if ((desc.idVendor  != TOMTOM_VENDOR_ID) ||
-        ((desc.idProduct != TOMTOM_MULTISPORT_PRODUCT_ID) &&
-         !IS_SPARK(desc.idProduct)))
+    if ((desc.idVendor  != TOMTOM_VENDOR_ID))
+//        ((desc.idProduct != TOMTOM_MULTISPORT_PRODUCT_ID) &&
+//         !IS_SPARK(desc.idProduct)))
     {
         *watch = 0;
         return TTWATCH_NotAWatch;
@@ -411,7 +411,7 @@ int ttwatch_open_device(libusb_device *device, const char *serial_or_name, TTWAT
     if (count > 0)
         ((char*)(*watch)->serial_number)[count] = 0;
 
-    RETURN_ERROR(ttwatch_send_startup_message_group(*watch));
+//    RETURN_ERROR(ttwatch_send_startup_message_group(*watch));
 
     // get the watch name
     if (ttwatch_get_watch_name(*watch, name, sizeof(name)) != TTWATCH_NoError)
[vmindru@vmindru ttwatch]# 
```

ryanbinns, 2017-03-17:
Yes, that will make it partially work, but only because you're disabling all the error-checking to make sure it's a watch you're talking to.

What you have is a new watch that has a unsupported product ID, and seems to communicate differently. It might use the new Spark protocol, so try applying this patch and have another go:
```
diff --git a/include/libttwatch.h b/include/libttwatch.h
index 18bc8e6..c9016c3 100755
--- a/include/libttwatch.h
+++ b/include/libttwatch.h
@@ -20,10 +20,12 @@ extern "C" {
 #define TOMTOM_MULTISPORT_PRODUCT_ID    (0x7474)
 #define TOMTOM_SPARK_MUSIC_PRODUCT_ID   (0x7475)
 #define TOMTOM_SPARK_CARDIO_PRODUCT_ID  (0x7477)
+#define TOMTOM_TOUCH_PRODUCT_ID         (0x7480)

 #define IS_SPARK(id)                            \
     (((id) == TOMTOM_SPARK_MUSIC_PRODUCT_ID) || \
-     ((id) == TOMTOM_SPARK_CARDIO_PRODUCT_ID))  \
+     ((id) == TOMTOM_SPARK_CARDIO_PRODUCT_ID) || \
+     ((id) == TOMTOM_TOUCH_PRODUCT_ID))  \

 /*****************************************************************************/
```

vmindru, 2017-03-17:
damn ... I spotted this yesterday while going trough the code.  However missed the part to add it to the 
IS_SPARK let me try that.

vmindru, 2017-03-17:
btw @ryanbinns  I think you are missing a parenthesis.

ryanbinns, 2017-03-17:
I don't think so. It compiled fine for me...

vmindru, 2017-03-17:
actually, you are right, I just patched "manually" and pasted something wrong. went just fine with git apply.  I feel we are one step closure :-D now the command just hangs for 20 seconds. 

```
[root@vmindru ~]# time ttwatch --devices                                                                  

real    0m20.029s
user    0m0.014s
sys     0m0.011s
[root@vmindru ~]# time ttwatch --devices

real    0m20.017s
user    0m0.003s
sys     0m0.008s
[root@vmindru ~]# time ttwatch -v
Unable to open watch

real    0m20.014s
user    0m0.003s
sys     0m0.006s
[root@vmindru ~]# time ttwatch -v
Unable to open watch

real    0m20.013s
user    0m0.003s
sys     0m0.006s
[root@vmindru ~]# 
```

vmindru, 2017-03-17:
let me pull out an strace, maybe it will help.

vmindru, 2017-03-17:
here is the strace with hanging command after your patch suggestion http://pastebin.com/wP9yRhLC

ryanbinns, 2017-03-17:
I've never tried debugging using strace before, so I'm not sure how to interpret that. The delay generally only happens if the watch is busy. Can you verify that no other program is trying to use it? No other lingering copies of `ttwatch` in the background. I assume you're not trying to run the daemon at the moment.

vmindru, 2017-03-17:
@ryanbinns i don't seem to run the daemon at least i did not start it, let me try to make sure nothing else would be polling it.

ryanbinns, 2017-03-17:
Try running `lsof | grep dev/bus/usb` to see if there is a process that's holding the watch open. You could try reinserting it also.

vmindru, 2017-03-17:
Ok I have disabled again error-cheking and we are back where we where, I guess your assumption that the watch is using the Spark protocol is not true :( cause the ttwatch still can not talk to TomTom Touch :( . 

```
[root@vmindru ttwatch]# ttwatch -v
PollingProduct ID:       0x00000000
BLE Version:      0
Firmware Version: 0.0.0
Unable to read watch name
```

vmindru, 2017-03-17:
btw at the runtime nothig else is opening the watch 

```
[root@vmindru ttwatch]# lsof | grep dev/bus/usb
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
ttwatch   17038          root    9u      CHR            189,143       0t0  138292980 /dev/bus/usb/002/016
ttwatch   17038 17039    root    9u      CHR            189,143       0t0  138292980 /dev/bus/usb/002/016
[root@vmindru ttwatch]#
[root@vmindru 002]# lsusb | grep TOM
Bus 002 Device 016: ID 1390:7480 TOMTOM B.V. 
[root@vmindru 002]# 
```

vmindru, 2017-03-17:
Let me try to contact support and see if they are willing to share some info on the protocol this wristband is using.

VM

ryanbinns, 2017-03-17:
Your `lsof` output indicates there is a copy of `ttwatch` running that has a watch open. Or at least, it did a short time ago. Sometimes `lsof` takes a while to realise the device is closed. Does anything show up if you run `ps -ef | grep ttwatch`.

No, they won't help. I had to reverse engineer the watch myself to work out how to talk to it.

vmindru, 2017-03-17:
@ryanbinns  bastards :(

vmindru, 2017-03-17:
@ryanbinns  btw thanks a lot for trying to help here, if you want  we can switch to IM not to pollute this thread.  do you use IRC? if yes i am under nick:vmindru on freenode 

```
[root@vmindru ~]# lsof | grep -i /dev/usb/
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
[root@vmindru ~]# ps ufax | grep ttwatch
root     27827  0.0  0.0 118488  2248 pts/8    S+   13:13   0:00                  \_ grep --color=auto ttwatch
[root@vmindru ~]# 
```

ryanbinns, 2017-03-17:
No, I don't use any IM.

It seems like this watch uses a different protocol, so we may have to do a packet sniff while the Windows software is talking to the watch. Do you have a Windows PC you can use to do this? I used Wireshark/USBPcap to dump the raw packets and then went through them by hand with a hex editor to decipher what was going on. Hopefully the basic protocol will be similar, but there may be some low-level differences (the original vs Spark watches had a different packet size but the rest was the same, for example).

vmindru, 2017-03-17:
I will have to find a windows box :( and try to discover this part. I will let you know when/if I have any data. 
Thank you for now!

DrYak, 2018-09-13:
Hello, did you guys managed to get the Tomtom Touch working with ttwatch eventually ?

brocciu, 2023-11-15:
Hello, I have a new watch (touch) and I have ttwatch working on my mac. I have the message "Unable to open watch" even with "sudo".
Do you have a solution to set up a new watch with ttwatch ? 
Thank you,