ryanbinns/ttwatch #124: ttwatch --set-name succeeds but return error "Unable to write new watch name"

Grimler91, 2017-03-29:
When updating the name of my watch, it reports that it fails, even though it doesn't:

    $ /usr/local/bin/ttwatch --devices
    **SOME_ID**: TomTom GPS Watch
    $ /usr/local/bin/ttwatch --set-name "Grimlers tomtom"
    Unable to write new watch name
    $ /usr/local/bin/ttwatch --devices
    **SOME_ID**:  Grimlers tomtom

Same phenomena with and without root. --set-time works fine. I did some digging and this is where the error comes from:
 
    if (ttwatch_update_preferences_modified_time(watch) != TTWATCH_NoError)
        write_log(1, "Unable to write new watch name\n");

at line 246 in `ttwatch.c`. 

The error code is 16, i.e. `TTWATCH_InvalidParameter` which should mean that error comes from line 992 in `libttwatch.cpp`:

    if (!watch)
        return TTWATCH_InvalidParameter;

I see that DAYNAMES and MONTHNAMES are defined as Sun, Mon, ... and Jan, Feb, ... in ttwatch_update_preferences_modified_time. The language of my TomTom Runner 2 Cardio+Music watch is set to Swedish which means that the day names and month names are different (I think), could this be the cause of the problem?

Please let me know if I can provide any more information!

Grimler91, 2017-03-30:
Problem has disappeared today (after rebooting computer). I'll reopen if it appears again.

closed 2017-03-30 by Grimler91

ryanbinns, 2017-03-31:
Interesting. Glad it's working now though. FYI, I don't think the language could have been the issue anyway, since those names are used only for updating the internal last-modified timestamp in the preferences XML file, which is always is English anyway.

Grimler91, 2017-03-31:
@ryanbinns Thanks for the information, good to know!