ryanbinns/ttwatch #128: Firmware 1.3.255, R/O no manifest file

petrkutalek, 2017-04-21:
Hi @ryanbinns, I would like to thank you for you efford and nice ttwatch code. I like it. However, my Runner 3 watch runs firmware version 1.3.255. Is there any way I could help you to generate manifest file for this firmware?

ryanbinns, 2017-04-26:
Generating the manifest is not difficult, but very, very time-consuming. The basic idea is to save a copy of the manifest file (ID 0x00850000), change _**one**_ setting on the watch, get another copy of the manifest file and see what changed between the two. And so on and so forth, for all possible settings and values for each setting. It took me a week or more the first time I did it. Since then I've mainly just looked for new settings and assumed the existing ones are correct unless someone tells me otherwise.

petrkutalek, 2017-04-26:
@ryanbinns OK, I am going to check it during this weekend and let you know. Please, stand by.

closed 2018-04-15 by petrkutalek