ryanbinns/ttwatch #131: ttwatch stopped working before really starting

teihla, 2017-08-03:
I am really new to ttwatch. I installed it yesterday without problems. I was able to download one activity from my watch to the computer (using the daemon). When I tried to upload the activity to mapmyfitness, I received errors regarding permissions. I thought that was because the activity file was owned by root. I deleted it, and restarted the daemon with me as a user, and nothing has worked since. I am unable to download any activities from the watch, even a new one from this morning. Any idea what I should do next?

ryanbinns, 2017-08-03:
It sounds like a permissions issue. Firstly, make sure there are no copies of `ttwatch` or `ttwatchd` running.
1. Does it work if you use the `ttwatch` program without using the daemon (`ttwatch -a`)?
2. If not, does it work if you run the `ttwatch` program as root (`sudo ttwatch -a`)?
3. If so, then try doing `make install_udev` to install `udev` rules to make the watch accessible to the `usb` group. It will also add your username to the `usb` group so that you have permissions to access it.
Try these first. If you still can't get it working, there may be something else we can try.

teihla, 2017-08-03:
Hi, thanks for your response.

No activity gets loaded even if I run ttwatch -a. The output is:

Downloading GPSQuickFix data file...
Writing file to watch...
GPSQuickFix data updated
Current firmware is already at latest version

I also installed the udev rules and added myself to the usb group yesterday
following the steps in the instructions, i.e., I did not run make
install_udev. Running it now gives the following errors:

addgroup: The group `usb' already exists and is not a system group. Exiting.
CMakeFiles/install_udev.dir/build.make:57: recipe for target
'CMakeFiles/install_udev' failed
make[3]: *** [CMakeFiles/install_udev] Error 1
CMakeFiles/Makefile2:324: recipe for target
'CMakeFiles/install_udev.dir/all' failed
make[2]: *** [CMakeFiles/install_udev.dir/all] Error 2
CMakeFiles/Makefile2:331: recipe for target
'CMakeFiles/install_udev.dir/rule' failed
make[1]: *** [CMakeFiles/install_udev.dir/rule] Error 2
Makefile:253: recipe for target 'install_udev' failed
make: *** [install_udev] Error 2

Like I said downloading the activity worked once (it did not get deleted
from the watch, though)

--tei

ryanbinns, 2017-08-03:
Ok, I think I understand. Most people get confused by this at first. When the activity is downloaded from the watch, the activity data is definitely deleted from the watch. However, the watch still retains a summary of the activity for use when looking at history and for the race function. So it still looks like it's there, but there is no more data that can be downloaded. If you run `ttwatch --list-history` you'll see the activity, but the information displayed there is the only information that's left on the watch. Everything else has been deleted.

If you do a new activity then try to download it, it should work.

teihla, 2017-08-03:
Ok, I see. So the activity data also gets deleted if I use TomTom MySports
mobile app to download the activity. I did a small test activity and the
download using ttwatch worked just fine.

But when I try to run the script ./ttbin2mysports, I get:

./ttbin2mysports: 62: cd: can't cd to /home/tei/ttwatch/2017-08-03/*/
{"error":"validation/username,password","error_description":"Post body
failed validation:
{\"username\":\"validation/string\",\"password\":\"validation/string\"}"}curl:
(26) couldn't open file "/home/tei/ttwatch/2017-08-03/*//*.tcx"

I guess the "cant' cd '' error is because there is no subdirectory under
~/ttwatch/2017-08/03, but the rest?

--tei

ryanbinns, 2017-08-03:
That's correct, the activity data is always deleted, no matter what method you use for downloading it.

I don't know if `ttbin2mysports` works any more. I didn't write it and have never used it because I don't use MySports. TomTom changed the way MySports works some time ago. It used to be run by MapMyFitness, but TomTom subsequently rewrote the website using their own API. I'm not sure if anyone else uses the `ttbin2mysports` script or if it still works.

BTW, did you change the username and password inside the script (lines 9 and 10)? If not, try changing them to your username and password and see if it works. You might be lucky.

teihla, 2017-08-03:
Yes, I did change the username and password. So, if the script does not
work, there is no way I can get the activity to MySports? That was the
whole point, since the mobile app is so unreliable)

--tei

ryanbinns, 2017-08-03:
Sorry no, you're the first person I have found to want it since the API was changed. Nobody else has mentioned it. Most people use Strava instead as it's much better and you can just upload TCX files through the website interface. MySports is extremely limited.

The MySports API is not documented so it would need some serious reverse engineering to work out (it's encrypted communications, which makes it difficult).

teihla, 2017-08-03:
Ok, thanks for the info.

I got the script working up to the point it generates the .tcx file and
*uploaded_to_mysports file, but it does not actually upload the file to
MySports, and I can't find any way to upload files on the website either,
just to download.

 I keep trying ...

--tei