ryanbinns/ttwatch #135: Add fitness points to ttbin decoding

scubajorgen, 2017-10-03:
@derickr Please add support for TomTom Fitness points to the ttbin decoding.
The new TomTom Fitnesspoints seem to be present in the ttbin record with tag 0x4A (The 0x4B record is the new variable length record.):

Format of the record:

1. Tag 0x4A (1 byte)
2. Timestamp (epoch seconds UTC) (4 bytes)
3. Fitness point counter (2 bytes)
4. Fitness point counter (2 bytes)

It is unclear to me what the difference is between the two fitness point values. They seem to have the same value. Both are counters are reset to 0 at 00:00 hours local time and increase with each workout.

The fitness points are also tracked in the 0x00B1nnnn files (activity tracking). I'll add a new protobuf definition later reflecting this extension.

derickr, 2017-10-03:
The question is however, what to do with the fitness point information? The TCX and GPX formats don't really do anything with this of course.

derickr, 2017-10-03:
Preliminary implementation in #136 .

ryanbinns, 2017-10-09:
Does the official TomTom software do anything with this data? If so, what?

scubajorgen, 2017-10-09:
For sure! Fitness points registered in the logged activities (ttbin files, 0x0091nnnn files) and in the tracked actitivity (0x00b100nn files). The former are displayed with each activity, the latter is displayed in a fancy new screen showing current weeks' fitnesspoints with stars (and new graphs with fitness age (yeah! this 50 year old nerd reached 20 years fitness age)).
[image]
[image]

ryanbinns, 2017-10-09:
I see. It looks pretty, but I'm not sure how we should implement this. `ttwatch` doesn't have a GUI, and none of the current output formats support adding this data to it. TCX could potentially be extended, since it's simply XML, but it might break other software that reads the files. A separate file could be created, but that could lead to issues with keeping the data together. Any ideas?

derickr, 2017-10-12:
As I said in #136 — I'll likely want to add a JSON output that has *all* the records in it.

jkopacze, 2017-10-16:
Hi guys, just to to let you know. The last change in TCX file format broke an integration with Endomondo, basically importing training file now throws an error about not being able to process  the file.

derickr, 2017-10-16:
@jkopacze That seems curious, as my changes did not modify the TCX output format. Can you share the ttbin and tcx files, and if you have, a better error message?