ryanbinns/ttwatch #137: TCX has wrong/strange laps

mipapo, 2017-10-06:
It's more a question, because I'm not sure if it's related to this repository or a TomTom problem.
Watch: TomTom Runner2 
Firmware: 1.6.26 (HN2485G-01020)

The converter generates a lot of laps which are wrong. Does anyone have an idea what the problem is?

TTBIN file
[Running_09-54-03 (1).ttbin.zip](https://github.com/ryanbinns/ttwatch/files/1361859/Running_09-54-03.1.ttbin.zip)
Generates:
[Running_09-54-03 (1).ttbin.zip](https://github.com/ryanbinns/ttwatch/files/1362397/Running_09-54-03.1.ttbin.zip)
[image]

mipapo, 2017-10-07:
Seems to be a bug in that specific TomTom firmware version. In 1.7.53 is everything fine

closed 2017-10-07 by mipapo

ryanbinns, 2017-10-09:
That's strange. Were there any similar problems reported in the official TomTom forums?

mipapo, 2017-10-09:
Seems it was not pushed to all devices and they had to skip the fw version (https://de.discussions.tomtom.com/spark-runner-2-spark-3-runner-3-475/version-1-6-26-988760)