ryanbinns/ttwatch #139: Wrong steps (count/length) in treadmill

OswaldKolle1977, 2017-10-26:
Don't know, if it is only with my runs, but I get a wrong Step-Count and -Length (in treadmill):
Values: 160 spm, 0,64m (step-count: 9616)
calculated: 8 spm, 2,55m (step-count: 481)

The failure is within the "export_tcx.c".
The Code from line 265 (with my correct working version):
```// BEGIN ORIGINAL
                if ((int)record->treadmill.steps <= 4) // max 4 * 60 = 240 spm
                    cadence_avg = (0.05 * 30 * (int)record->treadmill.steps) + (1.0 - 0.05) * cadence_avg;
                fprintf(file, "                                <RunCadence>%d</RunCadence>\r\n", (int)cadence_avg);
// END ORIGINAL
// BEGIN MW (WORKING)
//                if ((int)steps <= 4) // max 4 * 60 = 240 spm
//                    cadence_avg = (0.05 * 30 * (int)steps) + (1.0 - 0.05) * cadence_avg;
//                fprintf(file, "                                <RunCadence>%d</RunCadence>\r\n", (int)cadence_avg);
// END MW (WORKING)
 ```
It seems like the values within record->treadmill are not correct, but the steps are...

ryanbinns, 2017-10-27:
Yes, you're correct. I have pushed an update with the fix. Thanks for tracking this down. It's been there for a long time; I guess not many people use the treadmill function.

OswaldKolle1977, 2017-10-27:
Thanks a lot!

closed 2017-10-31 by ryanbinns