ryanbinns/ttwatch #140: Has TomTom firmware update broken ttwatch?

martindlux, 2017-10-30:
I can no longer transfer my activities using ttwatch:
`Warning: Corrupt or unrecognisable activity file: Unknown_13-35-42_2348.ttbin`

The activities do, however, appear correctly in the history list.

I think the device firmware was updated recently, could that be causing the problem?
`martind@bart:~> ttwatch -v
Product ID:       0xd3070000
BLE Version:      1030826
Firmware Version: 1.7.53
Watch Name:       martind
Serial Number:    HO6495G02037
`
I have tried re-compiling and re-installing ttwatch but this does not seem to help.

Any tips/pointers as to how to get things working again?

ryanbinns, 2017-10-31:
That's a little strange. I've just tried my watch and it worked fine. I only tried a short running activity though. What type of watch do you have? Are you happy to send me the `ttbin` file?

martindlux, 2017-10-31:
Sorry just realised that the newly complied ttwatch had not overwritten the earlier version. 

All seems to be ok again now.

ryanbinns, 2017-10-31:
Aah ok. That's good to hear.

closed 2017-10-31 by ryanbinns