ryanbinns/ttwatch #145: Documentation update required: Firmware 1.2.0 does not work with ttwatch (or is TomTomconnect needed at least once ?)

insensitiveclod, 2018-02-02:
Hello

I'd been given a Multi-sport/Runner (1390:7474) as a christmas present a long time ago. Sadly, at that time there were no ways to use the device with Linux.
Recently, I discovered ttwatch's existence and was delighted to know that there might be a way for me to use the watch anyway. 
After initially building the tool, the only thing I could ever get it to do was to print:
Unable to open watch

All the operations, --initial-setup, --update-fw, etc.. all printed this error.
Then, before doing more work on debugging the issue using strace and debug, I decided to try upgrade the firmware first. Sadly this didnt want to work via wine(-devel), but I got a friend with a windows-machine to install the TomTom connect software and run an update for me.

After this, the watch now works.
I think it would be *VERY* desirable to inform people that they might need to update the firmware on older watches before using it with ttwatch OR to inform them to at least connect it to a windows-machine once.
This cost me a good few hours of time to debug, without need..it'd seem ;)
jfyi

Now with 1.8.x firmware on the watch, it's communicating nicely with ttwatch. Havent done much, but it all seems to do *something*. I'm happy as a clam. Thanks for the software !
-- 
 Justa

mipapo, 2018-02-02:
Adding a hint in the Readme what be okay, but I don't think it is the task of ttwatch to inform users of the software that they should update the firmware of their sport watches.

insensitiveclod, 2018-02-02:
The issue , really, is that if you use linux exclusively (like I do), you may not have a way to update the firmware at all. It becomes a rather frustrating experience when ttwatch --fwupdate doesnt work....because your firmware is too old (and you dont even realize it).

I was lucky to have a friend around at work who didnt mind booting up his neglected win7 install on his laptop to let me do a check for new firmware before I started taking ttwatch apart in a debugger.

Now that ttwatch works for my watch, the firmware will be up-to-date from now on.....Before now, I just didnt have any means to do so. If this is a pre-requisite for ttwatch to work, it would be good to document that for new , unsuspecting, users.

ryanbinns, 2018-02-03:
I was not aware that it would not update the firmware for older watches. I've never had firmware before 1.8.x that I can remember, so I've never had the opportunity to test it.

I'll add a note in the readme file.

closed 2018-02-03 by ryanbinns