ryanbinns/ttwatch #146: can't sync local time?

katakombi, 2018-02-03:
thanks for ttwatch which seems to work just fine in general with my tom tom watch.
However, I can't seem to sync local time. When I run 

```
./ttwatch --get-time
UTC time:   2010-01-01 05:45:57
Local time: 2018-02-03 15:41:23 (UTC+70929.9)

./ttwatch --set-time
Watch has wrong UTC time. Use GPS to correct this.
```

Then I update the GPS fix
```
./ttwatch --update-gps
Writing file to watch...
GPSQuickFix data updated
```
but still no success
```
./ttwatch --set-time
Watch has wrong UTC time. Use GPS to correct this.
```

This is my device
```
./ttwatch -v
Product ID:       0xec030000
BLE Version:      28
Firmware Version: 1.8.52
Watch Name:       
Serial Number:    HE4304G01110
```

Maybe I am misinterpreting the --set-time option, but isn't it supposed to set the clock's time absolutely?

scubajorgen, 2018-02-03:
When you normally use the watch for GPS tracking it synchronizes to the satellite time. That's what meant by 'Use GPS to correct this'. Apparently you have factory reset the watch, since its time is set to 2010... Go outside, lock on the GPS and try again.
What ttwatch does is set a fixed offset, e.g. to 3600 seconds for UTC+1.

katakombi, 2018-02-04:
Hmm, thanks for the explanation. But then it is unclear to me what
```
./ttwatch --update-gps
```
does? I knew that I wouldn't get a GPS fix at home where I was running these commands, and I assumed that option would replace the GPS fix. If not, what else does it do?

ryanbinns, 2018-02-04:
When you use `--update-gps` it only updates the GPS satellite orbit information so that the watch can get a faster fix. It doesn't actually give the watch a GPS fix. When the watch gets a GPS fix, the watch updates it's GPS offset and it's internal time do that the visible time remains identical. When you sync time on the watch it updates the GPS offset so that the visible time is correct. It does not change ghee watch's internal time at all - that is always set to UTC time via GPS satellites. I hope this makes sense.

katakombi, 2018-02-04:
Thanks for your clarification, I guess I've got it now :)

closed 2018-02-04 by katakombi