ryanbinns/ttwatch #147: Systemd daemon configuration?

typhoon2099, 2018-02-11:
I'm having trouble with the ttwatchd service through systemd. I've got a running config but it runs as root so all of my activities are downloaded /root/ttwatch. I know there's a --runas flag but I can't see to get this to work properly. I've also tried setting up the service to run as my user (via a config in ~/.local/share/systemd/user) but again this doesn't work as the daemon needs to be run as root. Can someone please provide some help in getting it set up? Here's my current configuration:

```
[Unit]
Description=Starts ttwatchd for tomtom sport gps watch to sync activity files automatically

[Service]
Type=forking
ExecStart=/usr/local/bin/ttwatchd --auto

[Install]
WantedBy=graphical.target
```

ryanbinns, 2018-04-20:
Sorry, I must have missed this message. If it's still not working, I run the daemon as
```
ttwatchd --auto --runas <user>:usb
```
where `<user>` is my username. I haven't had any problems with it working, as long as the user has permissions to access the USB device. Have you ensured that the `ttwatch` program can run independently from the daemon as the required user?

typhoon2099, 2018-04-20:
I can indeed run `ttwatch` to get the activities from it. I currently have the daemon updating everything and I pull the activities manually. After my next run I'll try adding the --runas flag. Will I need to actually enter my username or can I use the %I (or whichever one it is) placeholder?

Reply to: reply@reply.github.com
Cc: github@matthewsimpson.net; author@noreply.github.com

typhoon2099, 2018-04-22:
I think it's working now. I had to set `--runas=<user>:usb` (note the equals). After a few restarts it seems to be downloading automatically. So I think this can be clised now.

typhoon2099, 2018-04-24:
Sorry, I've just tried to download a new activity this morning and I' getting the `Unable to write file` error in the logs and no activities downloaded. What permissions do you have set from the ttwatch directory and down? I've tried setting 777 permissions on all directories and creating the `2018-04-24` directory for ttwatchd to put the file in but there is still an issue downloading activities. `ttwatch -a` runs fine.

closed 2018-04-30 by typhoon2099