ryanbinns/ttwatch #148: [QUESTION] Playlists

angushendrick, 2018-03-02:
Thanks much for ttwatch. Is there any chance that support for putting playlists on the Spark 3 could be added?

ryanbinns, 2018-03-02:
That's a tough one. I don't have a Spark and am unlikely to get one, so someone else would have to provide details of how the playlists are transferred to the watch.

angushendrick, 2018-03-02:
I figured out an approach this morning that works. The watch is mounted as
a drive and I can manually put files on it. I'm not sure what mounts it,
but I have both the TomTom debian package and ttwatch installed and running
daemons. Thanks for following up.

scubajorgen, 2018-03-04:
Indeed, on my windows PC the storage for music is mounted as a removable storage.
Furthermore, file 0x00880000 contains a textual representation of the playlists. I'll add an example showing the format later.

typhoon2099, 2018-04-09:
I have tended to copy/paste songs into a new folder in /MySportsConnect/Music and then opened them all in Amarok (music player), then exporting the current playlist as an M3U8 and saving it in the directory. Then I need to update the master playlist with a link to the newly created playlist. It's a bit or work but it's doable. I'm sure the process could be automated somehow.