ryanbinns/ttwatch #150: Invalid conf file line: RunAsUser

typhoon2099, 2018-04-30:
I'm getting the following error when trying to load ttwatchd via systemctl. 

```
Apr 30 21:55:00 MJS-Envy ttwatchd[5704]: Invalid conf file line: RunAsUser=matthew:usb
```

This is the contents of `/etc/ttwatch.conf`:

```
RunAsUser=matthew:usb
GetActivities = true
GetActivities=true
UpdateFirmware=true
UpdateGPS=true
SetTime=true
Formats=csv,gpx,tcx
```

ryanbinns, 2018-05-17:
Try putting spaces around the '=' sign. Looking back at the code, it's a bit unclear and may be wrong.

typhoon2099, 2018-05-17:
I can't remember if that was the solution but I did eventually get this working.

typhoon2099, 2020-06-23:
Sorry to bring up an old post but I've reinstalled my distro and lost my daemon config, and now I'm back to having the daemon run but not have permissions to write files. `RunAsUser` fails with or without spaces, and using `--runas=matthew:usb` causes the permission errors. Can you post your exact `/etc/ttwatch.conf` and daemon files so I can see where I'm going wrong?

typhoon2099, 2020-06-23:
Looking through `src/options.c` I think there may be a bug in the RunAsUser option. The `global` int is initialised as `0` and, in the `RunAsUser` block (line 140) is changed to `result`, which only ever seems to be `0` at that point. Therefore I don't think the `RunAsUser` can ever be used properly.

ryanbinns, 2020-06-24:
Thanks for the report. Yep, it looks like it would never have worked. Strange. Anyway, I've pushed a patch which fixes the issue. I've tested it here and it starts as the RunAsUser user just fine. Let me know how you go.

typhoon2099, 2020-06-24:
`RunAsUser` is now accepted but I'm still getting write errors:
`/etc/ttwatch.d`:
```
GetActivities = true
UpdateFirmware = true
UpdateGPS = true
SetTime = true
Formats = tcx
RunAsUser = matthew:usb
Ephemeris7days = true
``

`/etc/systemd/system/ttwatch.service`:
```
[Unit]
Description=Starts ttwatchd for tomtom sport gps watch to sync activity files automatically

[Service]
Type=forking
ExecStart=/usr/local/bin/ttwatchd

[Install]
WantedBy=graphical.target
```

/var/log/ttwatch/ttwatch.log`:
```
Wed Jun 24 12:46:01 2020: Watch connected...
Wed Jun 24 12:46:04 2020: Unable to write file: Gym_22-13-19.ttbin
Wed Jun 24 12:46:04 2020: Unable to write file formats: tcx 
Wed Jun 24 12:46:05 2020: Downloading GPSQuickFix data file...
Wed Jun 24 12:46:05 2020: Writing file to watch...
Wed Jun 24 12:46:06 2020: GPSQuickFix data updated
Wed Jun 24 12:46:07 2020: Current firmware is already at latest version
Wed Jun 24 12:46:07 2020: Finished watch operations
```

My user is `matthew` and I have `usb` group permissions.

typhoon2099, 2020-07-11:
Ah ha! I needed to set the `ActivityStore` in the config. Sorted!

closed 2020-07-11 by typhoon2099