ryanbinns/ttwatch #151: gpx-file creation: order of longitude and latitude (lon / lan)

andreasma, 2018-10-18:
There are discussions of the correct order of the lon- and lan-attribute of the gpx files. ttwatch creates the file in the order lon / lan, but it should better done in the opposite. If you edit a gpx file (fetched from the TomTom watch) e.g. with gpsprune your edited file get the attributes in the opposite order. Thus you could not compare and join it with the original. Example: you have the heart rate in the original and want to combine it with the hight, that you imported from the internet into gpsprune and export it to a new gpx file.
I suggest to make a change to 'export_gpx.c' in line 58:
from:
record->gps.longitude, record->gps.latitude);
to:
record->gps.latitude, record->gps.longitude);

andreasma, 2018-10-19:
I reviewed the export_gpx.c file again and had to amend that the ordering is done a line before (line 57). This line has to be changed to:
fprintf(file, "            <trkpt lat=\"%.6f\" lon=\"%.6f\">\r\n",

ryanbinns, 2018-12-04:
Ok. I've made that change.

closed 2018-12-04 by ryanbinns