ryanbinns/ttwatch #154: Config option to download 7 day ephemeris instead of 3 ?

ABelliqueux, 2019-01-15:
Hi, 
Is there a configuration option to download the 7 days ephemeris instead of the default 3 days ?
IMHO, It would make sense to implement one if not or to switch to 7 days by default .
Right now, I have to rely on another software to upload the 7d ephemeris to the watch.
Maybe it is not as trivial as I think it is.
Cheers and thanks for this great software.

ryanbinns, 2019-01-17:
I have added a command line option to switch to a 7-day ephemeris (`-7` or `--eph7days`) and an associated config file option (`Ephemeris7Days = true`). This works for both the standalone program and the daemon. It is still 3-days by default.

Grimler91, 2019-02-11:
Does the watch use a 7 day ephemeris or does it "expire" after 3 days?
@dlenski experimented with it and reported that the watch only seemed to accept a 3 day ephemeris, see bottom of [tt_bluetooth.md Normal operation](https://github.com/dlenski/ttblue/blob/master/tt_bluetooth.md#normal-operation).

ryanbinns, 2019-02-11:
In my testing, it accepted the file and seemed to work properly. I haven't actually tested it to see if the GPS still locks quickly after more than 3 days, since I plug the watch in overnight and it automatically updates as soon as it's plugged in.

dlenski, 2019-02-11:
I no longer have a device to test with, but when I did I found that the original Runner v1 would sometimes crash after having a 7-day ephemeris file pushed to it. I found the whole thing surprisingly cranky and unreliable so I eventually settled on just mimicking the original firmware.