ryanbinns/ttwatch #156: Missing header file causes build fail

jrklasen, 2019-02-10:
Hi, at the latest master `make` complains about a missing header file.
```
$make
[  2%] Building C object CMakeFiles/libttbin.dir/src/export.c.o
In file included from /home/jrklasen/dev/ttwatch/include/protobuf.h:9:0,
                 from /home/jrklasen/dev/ttwatch/include/export.h:12,
                 from /home/jrklasen/dev/ttwatch/src/export.c:6:
/home/jrklasen/dev/ttwatch/src/protobuf/activity_tracking.pb-c.h:7:10: fatal error: protobuf-c/protobuf-c.h: No such file or directory
 #include <protobuf-c/protobuf-c.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/libttbin.dir/build.make:86: recipe for target 'CMakeFiles/libttbin.dir/src/export.c.o' failed
make[2]: *** [CMakeFiles/libttbin.dir/src/export.c.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/libttbin.dir/all' failed
make[1]: *** [CMakeFiles/libttbin.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

ryanbinns, 2019-02-11:
You have to make and install the Google protocol buffer library (https://github.com/protocolbuffers/protobuf) and the C-language binding for the protobuf library (https://github.com/protobuf-c/protobuf-c). I just realised the README.md file hasn't been updated to indicate this. I'll update it.

jrklasen, 2019-02-11:
thanks -- also for developing this great tool!

closed 2019-02-11 by jrklasen