ryanbinns/ttwatch #158: Does this work on a Mac

jacquesnoris, 2019-04-19:
Sorry, I'm a total newbie, TTwatch looks great, exactly what I was looking for, any chance to get this running (pun intended ;) on a Mac?

ryanbinns, 2019-04-25:
I don't have access to a Mac and I've never done any software development (and never really even used) a Mac, so I have no idea at all. It may be able to be done, but I won't be doing it.

uded, 2019-04-25:
Some time ago I had no problem compiling on Mac, but that was *really* a
while ago...

Pozdrawiam serdecznie & Best Regards & Mit freundlichen Grüßen,
-- 
:: *Łukasz Rżanek*

jacquesnoris, 2019-06-14:
Okay, thanks for your answers!

sanderevers, 2023-09-23:
Just compiled it on MacOS Monterey without problems. Installed the dependencies (cmake, openssl, curl, libusd, protobuf, protobuf-c) via Homebrew.
Only thing is that `ttwatch` only works using sudo (can't access the USB device otherwise).

Lenahic, 2023-10-05:
Hey !
What do you mean by "ttwatch only works using sudo" ? what are the instructions you wrote ?

Lenahic, 2023-10-05:
I guess I have managed to open the app using "sudo open TomTomWatch.jar", yet, the software doesn't detect the watch ...

sanderevers, 2023-10-06:
I don't know what `TomTomWatch.jar` is, but it has nothing to do with the `ttwatch` software in this repo. To install `ttwatch` on a Mac you first need to install the [Homebrew package manager](https://brew.sh/) and then use that to install the dependencies:
```
brew install cmake openssl curl libusd protobuf protobuf-c
```
Then
```
git clone https://github.com/ryanbinns/ttwatch
cd ttwatch
```
and follow the [build instructions](https://github.com/ryanbinns/ttwatch#build-instructions) which should install `ttwatch` into `/usr/local/bin`. You can test if that went right by just running
```
ttwatch
```
and see if it prints a help text.
If that works, you should be able to plug in your watch and do
```
sudo ttwatch --list-history
```
to see your latest activities on the watch. At least, this is what worked for me (on MacOS Monterey). Hope it does the trick for you too.

scubajorgen, 2023-10-06:
> @sanderevers : I guess I have managed to open the app using "sudo open TomTomWatch.jar", yet, the software doesn't detect the watch ...

Your confusing with my project: https://github.com/scubajorgen/TomTomWatch

Lenahic, 2023-10-06:
And i also have this error about the 'sprintf' function : 
ttwatch/src/libttwatch.cpp:1011:5: warning: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Wdeprecated-declarations]
    sprintf(timestr, "%s %d. %s %02d:%02d:%02d %04d",

sanderevers, 2023-10-06:
I don't know much about C, but my guess is it cannot find the `protobuf-c` headers. I noticed a typo in my `brew install` instructions (I wrote `libusd` instead of `libusb`), maybe that's the reason. Can you try `brew install cmake openssl curl libusb protobuf protobuf-c` again and do a new build?

Lenahic, 2023-10-06:
done but it doesn't change anything. Yet, running brew instal cmake etc ... I've some errors about the formula used : 

Warning: Treating cmake as a formula. For the cask, use homebrew/cask/cmake
Warning: cmake 3.27.6 is already installed and up-to-date.
To reinstall 3.27.6, run:
  brew reinstall cmake
openssl@3 3.1.2 is already installed but outdated (so it will be upgraded).
Warning: curl 8.3.0 is already installed and up-to-date.
To reinstall 8.3.0, run:
  brew reinstall curl
Warning: libusb 1.0.26 is already installed and up-to-date.
To reinstall 1.0.26, run:
  brew reinstall libusb
Warning: protobuf 24.3 is already installed and up-to-date.
To reinstall 24.3, run:
  brew reinstall protobuf
Warning: protobuf-c 1.4.1_7 is already installed and up-to-date.
To reinstall 1.4.1_7, run:
  brew reinstall protobuf-c
==> Downloading https://ghcr.io/v2/homebrew/core/openssl/3/manifests/3.1.3
############################################################################################################### 100.0%
==> Fetching openssl@3
==> Downloading https://ghcr.io/v2/homebrew/core/openssl/3/blobs/sha256:2cc86941823ff633c21ef3533d7a106c4bddbcb35adac8
############################################################################################################### 100.0%
==> Upgrading openssl
  3.1.2 -> 3.1.3 

==> Pouring openssl@3--3.1.3.ventura.bottle.tar.gz
==> Caveats
A CA file has been bootstrapped using certificates from the system
keychain. To add additional certificates, place .pem files in
  /usr/local/etc/openssl@3/certs

and run
  /usr/local/opt/openssl@3/bin/c_rehash
==> Summary
🍺  /usr/local/Cellar/openssl@3/3.1.3: 6,495 files, 30.0MB
==> Running `brew cleanup openssl@3`...
Disable this behaviour by setting HOMEBREW_NO_INSTALL_CLEANUP.
Hide these hints with HOMEBREW_NO_ENV_HINTS (see `man brew`).
Removing: /usr/local/Cellar/openssl@3/3.1.2... (6,495 files, 30.0MB)

It looks like it's fine but it's not ...

sanderevers, 2023-10-06:
Yes, your packages look fine. The `cmake` formula is also the one I have installed. I don't know what's wrong then...

Lenahic, 2023-10-06:
I managed to create all folders (ttbincnv, ttwatchd ...) manually and complete all the installation using "-i" to ignore errors. These folders are empty (normal ?)

Also, I complete the last instruction (sudo make install) and I found ttwatch into /usr/local/bin with all other files. All files previoulsy mentionned are still empty
When i try "ttwatch" instruction i've got "zsh: permission denied:"
And when i try : "sudo ttwatch" i've got "sudo: ttwatch: command not found"
:'(

sanderevers, 2023-12-20:
@Lenahic today I tried to install ttwatch on a clean MacOS Sonoma and got the same error. And I found the solution!
```
brew install pkg-config
```
After that, the build process worked.

sanderevers, 2023-12-20:
Also, to update the GPS data, I needed to create a file `/etc/ttwatch.conf` with the following line in it:
```
EphemerisURL=https://gpsquickfix.services.tomtom.com/fitness/sifgps.f2p{DAYS}enc.ee
```
I first tried to put this in `~/.ttwatch` but for some reason that doesn't work.