ryanbinns/ttwatch #159: Intervals wildly off

typhoon2099, 2019-04-25:
Some of my downloaded runs have crazy distances attributed to each lap in an interval run:

`<DistanceMeters>-296816.00</DistanceMeters>` and the like. As an example, [here's a link to a Strava activity](https://www.strava.com/activities/2299971800). As you can see, I've been to the sun and back!

All the other data about the run is fine, just the distances per lap is wrong (sometimes majorly in the negative, or 0m).

scubajorgen, 2019-04-25:
Fwiw:  I notice sometimes the TomTom ttbin files log a coordinate (0, 0) which lies in the sea west of Africa. You have to filter these out when processing, eg. when calculating distance. Otherwise you might get surprising results. 

⁣Sent from BlueMail ​

typhoon2099, 2019-04-25:
Interesting. I'll have a look through the TCX file but it went onto Strava and didn't have any unusual points on it. That said, Strava does a pretty good job of tidying up uploaded data (just not interval distances it seems).

typhoon2099, 2019-05-23:
Oh, I forgot to post the files for the run when I made opened this issue. Here's one from today.

[ttwatch.zip](https://github.com/ryanbinns/ttwatch/files/3213832/ttwatch.zip)

typhoon2099, 2019-08-15:
I've been digging into this issue today and I've found that in parse_ttbin_data if I look at p.record->interval_finish.total_distance then I get the crazy distances. However, I have been unable to track the issue back any further. Going by what I've read in the code I have to assume that the ttbin data itself is incorrect, as I can't find anywhere else where this value can be broken (or indeed where it was created in the first place). Whether or not ttwatch did something to the data or not I can't tell, nor do I want to try and debug this one run at a time (unless there's somewhere I can prevent the run from being removed from my watch). C++ is not my first language, so I could easily be missing something obvious here.

As an addendum I will point out that I have uploaded runs using the (absolutely awful) TomTom Sports app, and the intervals are correct there. They may just be ignoring the total_distance and calculating it using the difference in cumulative distances at the start and end of an interval.

PR #162 (2020-06-24): Use cumulative distance to calculate interval/lap distance
Reported distances are unreliable, this commit changes the code to calculate the interval distance based on the start/end cumulative distances.

Fixes #159
closed 2020-06-24 by ryanbinns