ryanbinns/ttwatch #160: ttwatchd not able to write activities and summaries

billie80, 2019-06-27:
First of all thanks for your great work on ttwatch. It is very much appreciated.

There is one issue though. I do not know what I am missing but ttwatchd is not able to write activities as well as summaries. Strangely ttwatch itself works as expected. I looked through the old issues related to problems with the daemon but found no clues.

Below some information which might help narrowing things down. I think I have everything set up as expected but maybe there is this one silly thing.

The device is detected:

> ~ $ dmesg
> hid-generic 0003:1390:7477.0021: hiddev96,hidraw3: USB HID v1.11 Device [TomTom TomTom GPS Watch] on usb-0000:00:14.0-13/input0

The device permissions also seem fine:

> $ ls -al /dev/hidraw3 /dev/tomtom
> crw-rw---- 1 root usb 246, 3 27. Jun 18:25 /dev/hidraw3
> lrwxrwxrwx 1 root root      7 27. Jun 19:56 /dev/tomtom -> hidraw3

However when connecting the watch the log shows:

> ~ $ tail -f /var/log/ttwatch/ttwatch.log
> Thu Jun 27 19:56:15 2019: Watch connected...
> Thu Jun 27 19:56:15 2019: Unable to write file: Pool_swim_19-37-24.ttbin
> Thu Jun 27 19:56:15 2019: Unable to write file formats: csv fit gpx kml pwx tcx 
> Thu Jun 27 19:56:16 2019: Unable to write file: Summary_2019-06-25.protobuf
> Thu Jun 27 19:56:16 2019: Unable to write file formats: csv 
> Thu Jun 27 19:56:16 2019: Downloading GPSQuickFix data file...
> Thu Jun 27 19:56:17 2019: Writing file to watch...
> Thu Jun 27 19:56:17 2019: GPSQuickFix data updated
> Thu Jun 27 19:56:17 2019: Current firmware is already at latest version
> Thu Jun 27 19:56:18 2019: Finished watch operations

The daemon is started as follows:

> ~ $ ps auxf | grep ttwatchd
> billie    8799  0.0  0.0  91124  9284 ?        Sl   18:24   0:00 /usr/bin/ttwatchd --auto --runas=billie:usb

The user billie is in the usb group:

> ~ $ groups billie
> lp wheel cdrom video cdrw usb users lpadmin portage

The permissions of the activity store location should be sufficient:

> ~ $ ls -al ttwatch/TomTom_Spark_3/
> drwxrwxr-x 7 billie usb 4096 27. Jun 18:38 .
> drwxrwxr-x 3 billie usb 4096 27. Jun 19:49 ..

As mentioned above running ttwatch successfully retrieves the data from the device. In this case the ttbin,tcx and csv files from the swim activity which failed for the daemon above:

> ~ $ ttwatch --auto
> Unable to write file formats: fit gpx kml pwx 
> Downloading GPSQuickFix data file...
> Writing file to watch...
> GPSQuickFix data updated
> Current firmware is already at latest version

billie80, 2019-06-27:
For debugging purposes I added `write_log(1, "Error code, errno = %d\n", errno);` in addition to the "Unable to write file" logging in src/get_activities.c to get some informations why fopen fails.

The error code returned is 13 (EACCES) meaning permission denied which is strange as the daemon runs with the correct permissions and the user:group has write permission to the activity store location.

ryanbinns, 2019-06-27:
What do you have in `/etc/ttwatch.conf`? The main difference between `ttwatch` and `ttwatchd` is that `ttwatch` loads a `ttwatch.conf` from your home directory, but `ttwatchd` does not. Make sure the activity store location is correct in `/etc/ttwatch.conf` as well as in `~/ttwatch.conf`. Or specify the activity store location on the command line of `ttwatchd`

billie80, 2019-06-28:
Thanks for your quick feedback. I thought you might be asking this and I should have provided this information in advance. To answer it I do not use a configuration file at all.

It is good that you are asking this anyway as it made me start thinking. What if the default directory the daemon tries to write to is not of the user running it but the user starting it.

So I added additional logging `write_log(1, "Trying to write to : %s\n", dir_name);` to the create_directory_name function in src/get_activities.c so it prints the name of the directory it tries to write the activity to.

It turned out that my assumption was true and it tried to write to /root/ttwatch/... . So the default uses the home directory of root who is starting the daemon as base and not the home directory of the user starting the daemon. After specifying the desired location explicitly on the command line as you suggested everything was fine.

Wouldn't it make more sense if the default activity store location is derived from the user running the daemon and not the one starting it? As you can see with the current way it is somewhat confusing. Don't know how hard it would be to implement this though.

ryanbinns, 2019-06-28:
Unfortunately, it's not that simple. If the daemon is run at system startup from an init script (such as how I've set up my system), then the initiating user is `root` and using `root`'s home directory would be pointless. It would be possible to read the desired user's starting home directory from `/etc/passwd` (using `getpwnam`), but that wouldn't work if the user has specified a different home directory in `$HOME`. So there isn't really any clean option other than specifying the full path in the config file or on the command line.

billie80, 2019-06-28:
I see. Then what about adding an additional note to README.md and/or the help section of the ttwatch daemon.

Maybe something like this might do:

Note: When starting the daemon with the `--runas` parameter the home directory for default activity store location `~/ttwatch` is still the home of the root user who is starting the daemon and not the home of the user running the daemon.
In this case the activity store location must be specified on the command line or in the configuration file `/etc/ttwatch.conf` pointing to a location where the unprivileged user has write access.

PR #161 (2019-07-09): Add an additional note to README.md
Clarify the default default activity store location in daemon mode. Fixes #160
closed 2019-07-09 by ryanbinns