ryanbinns/ttwatch #163: Bluetooth issues

kosnick, 2019-10-06:
Need to apologise before anything else, i think this is probably not the best place to ask for help.

As many others, i have issues pairing tomtom adventurer to the app on android. During pairing the watch would display the pin but the phone won't ask for it, keeps searching all the time.
Having read a bit on the tomtom forums, at the moment i believe that the watch somehow remembers the last pairing, with a different phone, and hence somehow it won't work correctly with a new phone - i base this assumption on the fact that somebody with that issue, 'forgot' the watch from the previous phone and then was able to pair again in the new one.
I do not have access to my old phone to try this solution.
Could it be that ttwatch could somehow inform the watch to be forgoten on the previous phone?
Again, just an assumption.
If anyone cares, thanks in advance

kosnick, 2019-10-06:
I have found also ttblue. Using it i was able to connect to my phone from a linux system. So my assumption above is not correct - i guess.
Still no connection to the phone app...

dlenski, 2020-01-05:
> Could it be that ttwatch could somehow inform the watch to be forgoten on the previous phone?

Yes, you should be able to tell the watch to forget previously-paired files by