ryanbinns/ttwatch #164: General question - is the reverse possible?

meowcat, 2019-10-14:
Hi
just because I think you are probably the best resource on the topic:
This project reads out TT watches to send the data e.g. to Strava.
Based on your knowledge, do you think it is feasible to make an IoT device that basically impersonates a TT watch, so it could upload an activity into the TT app / cloud?

To clarify, this device would be some Arduino with some sensor. It would generate .ttbin files, then I would connect it to the computer like a real TT watch and upload the data into the app, i.e. my device would need to understand the TT protocol.

To achieve this, we would need to 
* understand the protocol well enough so my device could answer all "questions" asked by the real software correctly
* know if there is any validation, e.g. if the TT cloud goes through steps to recognize that a device is genuine (which would be easy enough for them).

I would appreciate your input on this. For now this is just a project idea.

scubajorgen, 2019-10-14:
As I understand you right you want to create an IoT device that behaves like a real TomTom Watch and use the TomTom PC App (USB) (I guess not the mobile app (Bluetooth)) to get ttbin files from the 'Watch' and upload it in the TomTom cloud.
There might be two ways to proceed:

**Spoofing the watch (make it mimic a TomTom Watch)**
This encompasses following

1. Translate your sensor values to ttbin format on the IoT device
2. Mimic the watch on USB level, i.e. present it with the proper manufacturer ID and product ID, etc. I don't know if this is possible.
3. Implement the USB protocol

I believe this is a hard way to go. The protocols for 1. and 3. are on [my site](https://blog.studioblueplanet.net/?page_id=627) and of course in Ryans application. However, 2. will be the culprit I guess.

**Find a way to upload ttbin files to TomTom**

1. Translate your sensor values to ttbin format on the IoT device
2. Transfer the ttbin files to the PC and offer them to the TomTom App
3. Have the TomTom App upload the files to the back-end

To be honest, I do'nt know if there is a way to offer ttbin files to the TomTom App other than via the watch. I once had a clue that it should be possible (e.g. by double clicking the ttbin file or placing it in a directory,...). However, I did not succeed. An older version of the mobile app apparently did the job just by putting the ttbin files in a certain directory. Disabled apparently in most recent versions.

meowcat, 2019-10-14:
Hi,

thanks for the fast reply!

> I guess not the mobile app (Bluetooth)

Ideally, I would want to use the mobile app with Bluetooth, but I thought USB might be easier.

> However, 2. will be the culprit I guess, but I lack the knowledge how hard it is to spoof an USB device.

Yes, I have the same concerns, zero experience here. For Bluetooth probably even worse.

> The mobile app apparently does the job just by putting the ttbin files in a certain directory and starting the app. It uploads the ttbins.

So I could potentially just find a way to put ttbins into an Android folder e.g. by writing my own app that gets the files from my device, and puts them into the correct directory? That would be the easiest way, though I am not too familiar with all the permissions and sandboxing that Android does (or iPhone, for that matter). 

(Also, it's easy enough to try this out.)

dlenski, 2020-12-01:
> Ideally, I would want to use the mobile app with Bluetooth, but I thought USB might be easier.

The Bluetooth protocol (at least for v1 and v2 devices) is well-understood. See https://github.com/dlenski/ttblue and in particular [the Bluetooth protocol document](https://github.com/dlenski/ttblue/blob/master/tt_bluetooth.md).