ryanbinns/ttwatch #166: Downgrade firmware on TomTom Multisport

JakubkoM88, 2020-02-28:
Hello,
do you think that it would be possible to install older firmware version as it is installed in TT Multisport watch via your utilities?
I have one TT Multisport with firmware 1.8.52. But i'm thinking that this firmware version contains some bug, that drains battery in simple watch mode. My watch battery life in simple watch mode is same as in GPS Running mode - aprox 7h.
But for example battery life is on 80% after 25h in treadmil mode...
So i'm wondering if it will be posiblle somehow download older firmware version into my watch - to test if really firmware 1.8.52 contains bug (don't turn off gps sensor in watch mode?)
J.

scubajorgen, 2020-02-29:
Hi
My experience: I once executed a downgrade. It succeeded but resulted in a complete reset of the watch: all settings lost. The process I followed was simply to copy an older version of the firmware to the watch. You need to have such a file since TomTom only provides the latest afaik. 
Apparently it is not supported which means that you may brick your watch worst case... 
Jörgen 

⁣_______________________________________ 
                                       O 
          ><((°>                  o 
                                         <°)>< ​

JakubkoM88, 2020-02-29:
Hi,
thanks for quick reply. I found page with url to download older firmwares...
Url for info about 1.8.46 http://download.tomtom.com/sweet/fitness/Firmware/E9030000/FirmwareVersionConfigV2.xml...
To download Firmware file it is just http://download.tomtom.com/sweet/fitness/Firmware/E9030000/1_8_46/0x000000F0

I tried also url with 1_8_42 and works too...
Do you think this can be correct firmware?
And if yes, how to load file into watch?
Jakub

scubajorgen, 2020-02-29:
Beware. I don't know where you got this link from. I've had a multisports as well and the product ID was EC030000, not E9030000.
You should read out the proper product ID from your watch, using Ryans' software or mine (https://blog.studioblueplanet.net/?page_id=566) and use that to download the firmware. 

The preference file on your watch (file with ID 0x00F20000, XML format) contains the ConfigURL (in case of my Adventurer it is https://mysports.tomtom.com/service/config/config.json). 

This points to a page which includes "service:firmware":"https://sports.tomtom-static.com/downloads/firmware/{PRODUCT_ID}/FirmwareVersionConfigV2.xml"

FIll in the proper product ID (no 0x, all capitals) and you end up (for my Adventurer E0070000) with a firmware file list:
```
<FirmwareVersion>
  <latestVersion>
    <Major>1</Major>
    <Minor>7</Minor>
    <Build>64</Build>
  </latestVersion>
  <isCritical>no</isCritical>
  <URL>1_7_64/0x000000F0</URL>
</FirmwareVersion>
```

You find the firmware files by appending the URL to the same base dir. Again for my Adventurer

https://sports.tomtom-static.com/downloads/firmware/E0070000/1_7_64/0x000000F0

I did not know TomTom kept the old firmwares online...

Again, I don't know if downgrades are supported. Don't blame me if you bust your watch.

JakubkoM88, 2020-03-07:
Hi, i finally found some time to install and run Ryan's software.
My product ID is EA030000. I found 1_8_46 firmware on URL
http://download-test.tomtom.com/sweet/fitness/Firmware/EA030000/......
Now it comes the hardest part -  install Linux and use your project or give a chance to java for a first time?
I made a quick look into your code and after a modify of  creating a URL to file download - replace firmware version in fw_url with '1_8_46' and then simply use do_factory_reset can will force format method to read xml with info about 1_8_52 firmware, but download files for 1_8_46 version.
What do you think?
And don't worry, i will not blame you for anything what happens to my watch.
Jakub

scubajorgen, 2020-03-08:
You can use my software as-is. 
1. Install software
2. Modify the properties file: set debugMode to true. You get an extra menu by which you can upload files by hand and reset a.o.
3. Download firmware files from tomtom (i guess i use the same filename convention as tomtom) 
4. Upload by hand to the watch, one by one
5. Reset

I guess Ryans' software might have the same features
Jörgen 

⁣_______________________________________ 
                                       O 
          ><((°>                  o 
                                         <°)>< ​

JakubkoM88, 2020-03-09:
You are right, Ryans sowtware have Debbuging menu with Upload file option.
But when i tried upload 0x000000F0 file, i got error message file not found.
When i listed all files in watch(menu option in debugging) - in list there was not 0x000000F0.
Jakub

JakubkoM88, 2020-03-21:
Ou yes:) Problem was only with missing .bin extension:)
So i upload all files to watch(dont forget do download also firmware/EA030000/BLE/28/0x00000012). 
I check files order in Ryans app - first 0x00000012, 0x000000F0, 0x00010200, 0x00850000, 0x00850001, then the rest
And finally - my watch show 1.8.46 firmware version:)
Battery fully charged, i hope it will last longer than in 1.8.52.  
Jakub

voytas280se, 2024-08-07:
@scubajorgen 

Old firmware links are not working anymore, is there any chance You have those files?
I have performed a factory reset on my recently bought adventurer and it looks like now it's bricked waiting for firmware to be uploaded :(
`Product ID:       0xe0070000
BLE Version:      0
Firmware Version: 1.7.64
`

> You find the firmware files by appending the URL to the same base dir. Again for my Adventurer
> https://sports.tomtom-static.com/downloads/firmware/E0070000/1_7_64/0x000000F0
> 
> Note: the same link results in a list of 19 files for product ID EC030000 and E9030000:
> https://sports.tomtom-static.com/downloads/firmware/EC030000/FirmwareVersionConfigV2.xml

scubajorgen, 2024-08-07:
I got a couple of firmwares on github:
https://github.com/scubajorgen/TomTomWatch/tree/master/various/firmware

Beware, I might have made mistakes putting them there... Use at own risk

voytas280se, 2024-08-07:
>I got a couple of firmwares on github:
https://github.com/scubajorgen/TomTomWatch/tree/master/various/firmware
>
>Beware, I might have made mistakes putting them there... Use at own risk

I found it already and managed to bring my watch back to life with 1.7.64 :)
I found the same file in other sources as well (I compared checksums) but they were not marked by the E0070000 ID.
The only thing I had to do is to host locally similar structure to the one previously hosted by TT site and replace https with http for downloading the config.json file (I modified the java code).

The java tool is a little safer as it checks for the firmware files to be downloadable before wiping the watch.
Props for that!

EDIT:
I removed my previous comment about bricking the watch and have created a separate issue about the --factory-reset command.