ryanbinns/ttwatch #171: How to use the protobuf file generated by "--get-summaries" option ?

ABelliqueux, 2020-10-28:
Hi, 

When using the "--get-summaries" option, I get a .protobuf file. Is it supposed to be parsed by ttwatch or is it supposed to be parsed with a different program ?

Thank you for your reply.

ryanbinns, 2020-12-10:
I have an older watch which doesn't produce these files, so I'm sorry I can't help. I have no idea what they would be used for. There is nothing in the `ttwatch` program that can handle them; it merely saves them to disk.

scubajorgen, 2020-12-12:
> Hi,
> 
> When using the "--get-summaries" option, I get a .protobuf file. Is it supposed to be parsed by ttwatch or is it supposed to be parsed with a different program ?
> 
> Thank you for your reply.
I do 

Protobuf encoding is used to encode data on the TomTom watches especially for the more recent files as routes and activity tracking. Protobuf encoding is a very compressed format.

Usually you compile the _protobuf definition file_ (it's a .proto file) using a protobuf compiler. The compiler generates code in the language of your choice, for example java code. This code you can use to encode or decode protobuf messages (data objects to protobuf and vv). I guess you get a protobuf encoded message from the watch with --get-summaries. 

There are protobuf decompilers which can get the data, but they don't give a clue what the data means. They are usually used for reverse engineering. I used [protobuf inspector](https://github.com/mildsunrise/protobuf-inspector) to reverse engineer the TomTom files.

Rever to [my program for the protobuf definition files](https://github.com/scubajorgen/TomTomWatch/tree/master/various/protobuf) (.proto files)