ryanbinns/ttwatch #174: make error

dancesWithCycles, 2021-04-14:
Hi folks,
thank you so much for creating and maintaining this repository. I would like to build and use it. This is what I did so far.
```
mkdir build
cd build/
cmake ..
make
```
And this is the error I end up with.
```
~/GitHub/ttwatch/build$ make
[ 25%] Built target libttbin
[ 27%] Linking C executable ttbincnv
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `root_container__get_packed_size':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:20: undefined reference to `protobuf_c_message_get_packed_size'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `root_container__pack':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:27: undefined reference to `protobuf_c_message_pack'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `root_container__pack_to_buffer':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:34: undefined reference to `protobuf_c_message_pack_to_buffer'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `root_container__unpack':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:43: undefined reference to `protobuf_c_message_unpack'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `root_container__free_unpacked':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:53: undefined reference to `protobuf_c_message_free_unpacked'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `metadata__get_packed_size':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:65: undefined reference to `protobuf_c_message_get_packed_size'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `metadata__pack':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:72: undefined reference to `protobuf_c_message_pack'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `metadata__pack_to_buffer':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:79: undefined reference to `protobuf_c_message_pack_to_buffer'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `metadata__unpack':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:88: undefined reference to `protobuf_c_message_unpack'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `metadata__free_unpacked':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:98: undefined reference to `protobuf_c_message_free_unpacked'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `data_container__get_packed_size':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:110: undefined reference to `protobuf_c_message_get_packed_size'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `data_container__pack':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:117: undefined reference to `protobuf_c_message_pack'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `data_container__pack_to_buffer':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:124: undefined reference to `protobuf_c_message_pack_to_buffer'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `data_container__unpack':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:133: undefined reference to `protobuf_c_message_unpack'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `data_container__free_unpacked':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:143: undefined reference to `protobuf_c_message_free_unpacked'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `sub_data_container__get_packed_size':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:155: undefined reference to `protobuf_c_message_get_packed_size'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `sub_data_container__pack':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:162: undefined reference to `protobuf_c_message_pack'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `sub_data_container__pack_to_buffer':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:169: undefined reference to `protobuf_c_message_pack_to_buffer'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `sub_data_container__unpack':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:178: undefined reference to `protobuf_c_message_unpack'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `sub_data_container__free_unpacked':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:188: undefined reference to `protobuf_c_message_free_unpacked'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `device_info__get_packed_size':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:200: undefined reference to `protobuf_c_message_get_packed_size'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `device_info__pack':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:207: undefined reference to `protobuf_c_message_pack'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `device_info__pack_to_buffer':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:214: undefined reference to `protobuf_c_message_pack_to_buffer'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `device_info__unpack':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:223: undefined reference to `protobuf_c_message_unpack'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `device_info__free_unpacked':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:233: undefined reference to `protobuf_c_message_free_unpacked'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `software_version__get_packed_size':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:245: undefined reference to `protobuf_c_message_get_packed_size'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `software_version__pack':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:252: undefined reference to `protobuf_c_message_pack'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `software_version__pack_to_buffer':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:259: undefined reference to `protobuf_c_message_pack_to_buffer'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `software_version__unpack':
[... 71 lines cut ...]
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `record4__free_unpacked':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:593: undefined reference to `protobuf_c_message_free_unpacked'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `activity__get_packed_size':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:605: undefined reference to `protobuf_c_message_get_packed_size'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `activity__pack':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:612: undefined reference to `protobuf_c_message_pack'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `activity__pack_to_buffer':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:619: undefined reference to `protobuf_c_message_pack_to_buffer'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `activity__unpack':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:628: undefined reference to `protobuf_c_message_unpack'
/usr/bin/ld: libttbin.a(activity_tracking.pb-c.c.o): in function `activity__free_unpacked':
/home/user/GitHub/ttwatch/src/protobuf/activity_tracking.pb-c.c:638: undefined reference to `protobuf_c_message_free_unpacked'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/ttbincnv.dir/build.make:86: ttbincnv] Error 1
make[1]: *** [CMakeFiles/Makefile2:142: CMakeFiles/ttbincnv.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
I appreciate any hint that is pointing me in the right direction. I read the documentation and tried to install all dependencies on the following computer I am using.
```
~/GitHub/ttwatch/build$ uname -a
Linux tpx200 4.19.0-11-amd64 #1 SMP Debian 4.19.146-1 (2020-09-17) x86_64 GNU/Linux
```
 Hopefully I am just blind and do not see the issue myself. Please unblind me. ;-)

Cheers!

dancesWithCycles, 2021-04-14:
Hi folks,
I still do not now why, however after I build and installed protobuf c compiler from [this](https://github.com/protobuf-c/protobuf-c) repository on my own, also ttwatch ran entirely through the `make` call.

I am only guessing, but it might be related to the installation directory of protobuf c compiler. I was playing around and ended up installing it to `/usr/lib` instead of `/usr/local/lib`.

closed 2021-04-14 by dancesWithCycles

dg0yt, 2021-04-14:
> Please unblind me. ;-)

Well, just give some time for answers. The README says:

> Prebuilt packages should be available for most systems using the system's built in package manager (dpkg, yum, apt, rpm etc...). Make sure that the -dev version of the packages (eg. libssl-dev, libcurl-dev, libusb-1.0-0-dev) are installed so that the headers are available.

Since you seem to be using a variant of Debian, you would probably need to do:
~~~
apt-get install libprotobuf-dev
~~~

> I am only guessing, but it might be related to the installation directory of protobuf c compiler. I was playing around and ended up installing it to /usr/lib instead of /usr/local/lib.

Sure. If you compile yourself, you don't want this to mix random versions of libraries with (Debian) distribution libraries. That's why you would install your custom protobuf to `/usr/local`. And then you would tell cmake that it needs to look there, in general, or at least for protobuf.
Now you installed to `/usr/lib`, and possibly replaced files of Debian package libprotobuf. So other programs from Debian which rely on the specific version provided by Debian may fail. In the best case with a big bang. In the worst case in very subtle ways.

PtyMatt, 2023-09-24:
For the record, "Prebuilt package" of protobuf gives the error above on Debian/Ubuntu systems, but it's ok on Fedora.