ryanbinns/ttwatch #178: Tom Tom Sports discontinuation

kubagruszka, 2023-09-04:
I'm TomTom watch owner and by surprise I was informed this week that they decided to shutdown their MySports platform making their watches broken. I was looking for some replacement and came across this project but from the description I don't fully understand if it requires the MySports platform or it's completely offline ? If it will continue to work maybe worth to mention this in the README file for wider audience ? I'm interested to keep my watch working so happy to contribute if so.

https://help.tomtom.com/hc/en-us/articles/11748276052370-TomTom-Sports-discontinuation

c-a-p-666, 2023-09-04:
Hi! Does work fine without MySports, just switched to ttwatch + Trainingpeaks solution last week.
My workflow: download activities using ttwatchd + config-file as .gpx files and import manually to Trainingpeaks. Other applications with import funcionality (e.g. GarminConnect,...) should work as well as you can specify the output file format. Does even work better than TomTom MySports imo --> saved my watch :)

ltguillaume, 2023-09-04:
@c-a-p-666 Sounds good, but what do you do about missing GPS data? https://github.com/ryanbinns/ttwatch/issues/176

c-a-p-666, 2023-09-04:
--update-gps works fine with the watch, does not seem to come from tomtom but standard available GPS data. Can anybody confirm?

ltguillaume, 2023-09-04:
There seeems to be a data url for the GPS data in the watch's configuration. ttwatch uses it to update the GPS data, so it will probably stop working.

c-a-p-666, 2023-09-04:
Should be possible to identify - but even _if_ it is hosted by tomtom it should be a generic service for all their devices, it would just make no sense to host a file specifically for the watch as this is standard GPS information.
Even if it would not update the watch is still usable, it would only take longer to sync with the sattelites if not used for a while.

c-a-p-666, 2023-09-04:
Looks like generic tomtom service: 
https://www.tomtomforums.com/threads/problems-with-the-newest-map-update.33385/
and
https://help.tomtom.com/hc/en-gb/articles/360013960459-About-QuickGPSfix
"Navigation device" should refer to the used chipset instead of products.

ltguillaume, 2023-09-04:
If the files/urls are chipset-specific, I'd expect them to disappear for their watches eventually.

jdugge, 2023-09-14:
The URLs are not chipset specific, but since they are in a subdirectory called "fitness", it still seems possible that they will disappear at the end of September 2023:

http://gpsquickfix.services.tomtom.com/fitness/sifgps.f2p3enc.ee (3-day extended ephemeris)
http://gpsquickfix.services.tomtom.com/fitness/sifgps.f2p7enc.ee (7-day extended ephemeris)

ltguillaume, 2023-09-19:
[image]

snooppr, 2023-11-05:
First of all I want to express my gratitude @ryanbinns !

Thanks to his repository, I can continue to use my TomTom runer cardio watch (first model). In addition, I recently disassembled the watch and replaced the battery with a larger one (I barely fit everything in, but now the watch lasts almost 1.5 times longer than when I purchased it new).

I have a question: GPS update does not work:
`ttwatch --update-gps`
>GPSQuickFix data URL option EphemerisURL not configured

The patch is available here:
https://github.com/ryanbinns/ttwatch/commit/ec1a276455c165c316732dd4a2d85f658945d303

And the data from the link is still downloading:
http://gpsquickfix.services.tomtom.com/fitness/sifgps.f2p3enc.ee

But the update doesn’t work, you shouldn’t expect it, will it be fixed somehow in the near future?

ryanbinns, 2023-11-06:
@ltguillaume No. I haven't had a TomTom watch for some years since mine died, so there's not a lot I can do other than merge pull requests. I can't do any testing.

PtyMatt, 2023-11-07:
@ltguillaume with https://github.com/ryanbinns/ttwatch/pull/181 you can configure the QuickGPS url, "just" need to add it as a new config (as described in the readme).

Pull the repo and rebuild the project!