ryanbinns/ttwatch #183: "ERROR: USB Error: An interface is already claimed" when running TomTomWatch

Skylarkin, 2023-11-05:
I am running Arch-linux, latest kernel, everything updated. I suspect this has something to do with ttwatch because I tried running TomTomWatch before testing ttwatch and the software TomTomWatch **was** working, the device (TomTom Adventurer) was recognised with correct serial no. HL4486G01240 and updating Quick GPS worked.

OpenJDK Java 21 full runtime environment installed. I have tried uninstalling ttwatch and modifying udev rules back and forth with various alternatives, rebooting system, running as both user and root and also tried the Zadig WinUSB driver in Windows 11 but no luck, I get the same error message in Windows even after installing the Zadig driver according to instructions found here:

https://blog.studioblueplanet.net/software/tomtomwatch

I never tried running TomTomWatch on Windows before I started using it on Arch-linux so it was first after the problem arose on linux that I also tried to run TomTomWatch on Windows.

The command: ttwatch --initial-setup is supposed to create a default XML preferences file on the watch, don't know if running this command might have something to do with this issue. I cannot see any XML preference file on the watch when mounted though:

https://github.com/ryanbinns/ttwatch#initial-setup

ERROR: No valid device found
ERROR: Error retrieving firmware version
ERROR: USB claim exception during release: No interface is claimed
ERROR: No valid device found
ERROR: Error retrieving product ID
ERROR: USB claim exception during release: No interface is claimed
ERROR: USB Error: An interface is already claimed

#usb-devices:

T: Bus=01 Lev=01 Prnt=01 Port=01 Cnt=01 Dev#= 4 Spd=480 MxCh= 0
D: Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs= 1
P: Vendor=1390 ProdID=7475 Rev=01.00
S: Manufacturer=TomTom
S: Product=TomTom GPS Music Watch
S: SerialNumber=HL4486G01240
C: #Ifs= 2 Cfg#= 1 Atr=c0 MxPwr=100mA
I: If#= 0 Alt= 0 #EPs= 2 Cls=03(HID ) Sub=00 Prot=00 Driver=(none)
E: Ad=02(O) Atr=03(Int.) MxPS= 256 Ivl=125us
E: Ad=81(I) Atr=03(Int.) MxPS= 256 Ivl=125us
I: If#= 1 Alt= 0 #EPs= 2 Cls=08(stor.) Sub=06 Prot=50 Driver=usb-storage
E: Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E: Ad=84(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms

PtyMatt, 2023-11-07:
linked to https://github.com/scubajorgen/TomTomWatch/issues/43 (closed...)

Skylarkin, 2023-11-08:
I wrote a new udev rule to solve this issue, **99-driver-prevent.rules**. (Directory /usr/lib/udev/rules.d/).

**99-driver-prevent.rules**:

> SUBSYSTEMS=="pci", DRIVERS=="xhci_hcd", ATTRS{subsystem_device}=="0x1b21", ATTRS{subsystem_vendor}=="0x1043", ACTION=="add", ATTR{authorized}="0"

My **99-tomtom.rules**:

> SUBSYSTEM=="usb", ATTR{idVendor}=="1390", ATTR{idProduct}=="7475", SYMLINK+="tomtom", GROUP="users", MODE="0666"

After adding a new udev rule, reload the rules by running:

$ sudo udevadm control --reload-rules

This is working for me on my system (Arch-linux). I have to disable the **99-driver-prevent.rules**  if using ttwatch though, otherwise the software won't work.

jonath35, 2023-11-16:
This problem is really hard to detect. I don't even know how to clean my system from all the ttwatch stuff

Skylarkin, 2023-11-19:
It is very strange indeed, because I know the software TomTomWatch was recognizing the watch correctly before i tried ttwatch. It was the first time i tried TomTomWatch though since I didn't have the need to use it before the discontinuation of TomTom's online service and software so unfortunately I haven't been able to use the software fully, uploading activities etc. I am using AUR packages on my system since I run Arch-linux and the process of uninstalling an AUR package should remove everything from the system so for me this is confusing. Could there still be something left that conflicts? By the way, I am using an ASUS laptop with several usb ports, some usb 2.0 and one usb 3.0, I have tried all of them, same result. I wonder if ttwatch modifies anything on tthe watch itself, after running the command ttwatch --initial-setup, but from what I can see when looking at the files that are on the watch, nothing seems to be changed.

dhaiducek, 2023-12-07:
I haven't pinned down the exact configuration, but in this block, I did the following just now:
- Added a watch name
- Added `<ephemerisModified>0</ephemerisModified>`
- Deleted all but the `ConfigURL` 
Then I ran `ttwatch --initial-setup` again, and things appeared to start working in TomTomWatch:
https://github.com/ryanbinns/ttwatch/blob/b95fd2c3bfa2dfa95e406f4c360ada9c14af7451/src/libttwatch.cpp#L924-L936

So for me it did seem to be something related to that preferences file (and it could be something as simple as just the watch name, but I haven't tested bringing back the other configurations yet to see...)