ryanbinns/ttwatch #188: Building issue / undefined reference to libusb

joppez, 2023-12-13:
When I tried to build ttwach, I get the following errors after the 'make' command. The cmake . command works and gives no errors. Is there any solution to this? I'm try to install it on a Raspberry Pi 3b+.

--------------

pi@raspberrypi:~/ttwatch $ cmake .
-- Enabled daemon function
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/arm-linux-gnueabihf/libusb.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/ttwatch
pi@raspberrypi:~/ttwatch $ make
[ 34%] Built target libttbin
[ 34%] Built target libttwatch
[ 34%] Linking CXX executable ttwatchd
/usr/bin/ld: CMakeFiles/ttwatchd.dir/src/ttwatchd.c.o: in function `register_callback':
/home/pi/ttwatch/src/ttwatchd.c:178: undefined reference to `libusb_hotplug_register_callback'
/usr/bin/ld: CMakeFiles/ttwatchd.dir/src/ttwatchd.c.o: in function `main':
/home/pi/ttwatch/src/ttwatchd.c:353: undefined reference to `libusb_init'
/usr/bin/ld: /home/pi/ttwatch/src/ttwatchd.c:356: undefined reference to `libusb_has_capability'
/usr/bin/ld: /home/pi/ttwatch/src/ttwatchd.c:359: undefined reference to `libusb_exit'
/usr/bin/ld: /home/pi/ttwatch/src/ttwatchd.c:368: undefined reference to `libusb_exit'
/usr/bin/ld: /home/pi/ttwatch/src/ttwatchd.c:377: undefined reference to `libusb_handle_events_completed'
/usr/bin/ld: libttwatch.a(libttwatch.cpp.o): in function `send_packet(TTWATCH*, unsigned char, unsigned char, unsigned char const*, unsigned char, unsigned char*)':
/home/pi/ttwatch/src/libttwatch.cpp:245: undefined reference to `libusb_interrupt_transfer'
/usr/bin/ld: /home/pi/ttwatch/src/libttwatch.cpp:256: undefined reference to `libusb_interrupt_transfer'
/usr/bin/ld: libttwatch.a(libttwatch.cpp.o): in function `ttwatch_enumerate_devices':
/home/pi/ttwatch/src/libttwatch.cpp:311: undefined reference to `libusb_get_device_list'
/usr/bin/ld: /home/pi/ttwatch/src/libttwatch.cpp:324: undefined reference to `libusb_free_device_list'
/usr/bin/ld: libttwatch.a(libttwatch.cpp.o): in function `ttwatch_open':
/home/pi/ttwatch/src/libttwatch.cpp:332: undefined reference to `libusb_get_device_list'
/usr/bin/ld: /home/pi/ttwatch/src/libttwatch.cpp:339: undefined reference to `libusb_free_device_list'
/usr/bin/ld: libttwatch.a(libttwatch.cpp.o): in function `ttwatch_open_device':
/home/pi/ttwatch/src/libttwatch.cpp:355: undefined reference to `libusb_get_device_descriptor'
/usr/bin/ld: /home/pi/ttwatch/src/libttwatch.cpp:368: undefined reference to `libusb_open'
/usr/bin/ld: /home/pi/ttwatch/src/libttwatch.cpp:383: undefined reference to `libusb_kernel_driver_active'
/usr/bin/ld: /home/pi/ttwatch/src/libttwatch.cpp:385: undefined reference to `libusb_detach_kernel_driver'
/usr/bin/ld: /home/pi/ttwatch/src/libttwatch.cpp:388: undefined reference to `libusb_claim_interface'
/usr/bin/ld: /home/pi/ttwatch/src/libttwatch.cpp:411: undefined reference to `libusb_get_string_descriptor_ascii'
/usr/bin/ld: libttwatch.a(libttwatch.cpp.o): in function `ttwatch_close':
/home/pi/ttwatch/src/libttwatch.cpp:452: undefined reference to `libusb_release_interface'
/usr/bin/ld: /home/pi/ttwatch/src/libttwatch.cpp:455: undefined reference to `libusb_attach_kernel_driver'
/usr/bin/ld: /home/pi/ttwatch/src/libttwatch.cpp:456: undefined reference to `libusb_close'
/usr/bin/ld: libttwatch.a(libttwatch.cpp.o): in function `ttwatch_get_serial_number':
/home/pi/ttwatch/src/libttwatch.cpp:771: undefined reference to `libusb_get_device'
/usr/bin/ld: /home/pi/ttwatch/src/libttwatch.cpp:771: undefined reference to `libusb_get_device_descriptor'
/usr/bin/ld: /home/pi/ttwatch/src/libttwatch.cpp:773: undefined reference to `libusb_get_string_descriptor_ascii'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/ttwatchd.dir/build.make:210: ttwatchd] Fout 1
make[1]: *** [CMakeFiles/Makefile2:116: CMakeFiles/ttwatchd.dir/all] Fout 2

annkvth, 2024-01-03:
I have a similar (the same?) issue on Ubuntu 22.04. 
I have installed libusb libusb-1.0-0-dev from ubuntu repos. 
It is found by cmake, with the output 
`-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib/x86_64-linux-gnu/libusb.so
`

However, make fails at "Linking CXX executable ttwatch
/usr/bin/ld: CMakeFiles/ttwatch.dir/src/ttwatch.c.o: in function `main':"
with undefined references to libusb_init , libusb_exit, libusb_get_device_list , and so on.

JaroslawJarmolowicz-TomTom, 2024-01-04:
I installed libusb-dev because it was not compiling. With bots libusb-dev and libusb-1.0-dev compiles and works fine.

annkvth, 2024-01-04:
> I installed libusb-dev because it was not compiling. With bots libusb-dev and libusb-1.0-dev compiles and works fine.

That did the trick for me too, thanks!

closed 2024-01-07 by ryanbinns