ryanbinns/ttwatch #192: Bricked watch after using --factory-reset command

voytas280se, 2024-08-07:
The factory reset command first wipes the watch and then tries to download new firmware.
It's obviously not working anymore and leaves the watch in wiped state.

I managed to find the firmware somewhere and upload it from locally running http server but still: it would be nice to either first check if the files are present and then wipe out the watch OR just disable this option completely so that nobody gets hurt.

traloph, 2024-08-09:
Mind sharing where you got the firmware from, or pointers on how to find it?

scubajorgen, 2024-08-09:
From TomTom, while updating the watch... 

⁣Get BlueMail for Android ​

traloph, 2024-08-09:
Great, i see you have the firmware file saved in your repo, thanks! 
Tomtom doesn't serve them anymore afaik. Just have to figure out the flashing from the self-hosted file.

prynhart, 2025-03-18:
Hi @voytas280se - I've just secondhand purchased a watch (a Tomtom Spark), and unfortunately the seller must have undertaken a factory reset like this.  Is there any chance you might be able to help me ?  Unsure where to get the firmware or how to go about loading it onto the watch :-/

voytas280se, 2025-03-19:
I took the firmware from this great repository:
https://github.com/scubajorgen/TomTomWatch/tree/master/various/firmware

There's no easy way of loading it to watch unless You are capable of modifying existing tools to read different firmware location.